simondankelmann/SubMarine #1: APK

Lelexx59, 2023-02-23:
Hey, thanks for your work.

Would it be possible to publish the compiled .apk file?

I got an error on my phone when I try to install the apk compiled (release and debug) with gradle.

Thank you and sorry for my bad english.

simondankelmann, 2023-02-24:
Hi, maybe you can describe the Error a bit more Detailed, or try running the App directly from within Android studio. I assume If i send you a prebuilt apk the Error will be the Same. Can you tell me which Android Version you are using ?

Lelexx59, 2023-02-24:
Hi, 

Well I tried to install the release version and I got this error: The application was not installed because the package appears to be invalid.

And for the debug version I just got the message "Application not installed."

I did this test on a Xiaomi 11T
MIUI Global version 13.0.3.0
Android version 12 SP1A.210812.016

I'm not familiar with Android Studio and Graddle, maybe it's a "logical" problem, I succeeded to compile the application by following some tutorials on graddle but the subject seems so vast.

Sorry for my bad english.

Thanks

simondankelmann, 2023-02-24:
I generated APKs for App and Debug, hope it helps :)
let me know if theres is any success with that:

https://github.com/simondankelmann/SubMarine/tree/main/Android/Apk

Lelexx59, 2023-02-24:
I succeeded to install the application with Android Studio and Wifi debugging so I could try the application.

The release version you shared installs and works too.

Thanks for your work.

closed 2023-02-24 by Lelexx59