simondankelmann/SubMarine #4: Built it however android 13 on S22 Ultra

computerfreek274, 2023-05-06:
App crash I can't click anything it will just close the app if i even click buletooth button crash menu crash. If there any tips to getting this working ?

simondankelmann, 2023-05-07:
Please try to grant all permissions manually to the app

computerfreek274, 2023-05-09:
Still can't get it here is a screenshot video of what is happening https://youtube.com/shorts/m57X0kH4FD4?feature=share 

Is the SD card needed in this version I tried wiring it with and without it. 
And do you have the pinouts exactly for an ESP 32 wemos with the new version of the cc1101  with 8 pins ? I don't think I wired it wrong but you never know maybe that is the cause of it.

simondankelmann, 2023-05-10:
Hi, thanks for the Video :)
first of all, the sd card is not needed in this setup. 
regarding the pinout: @General-Zod1 updated the Readme.md with some pin layouts maybe this helps.
you could try to see whats written in the arduino serial monitor, if the cc1101 cant be connected there it should be logged, but that would not crash the app.
maybe you could run the app on your phone using android studio and share the exact error that occurs ?

d4ve10, 2023-05-10:
I also had the same problem using Android 13 on the OnePlus 8 while testing the app. The error message from Android Studio is
`android.database.sqlite.SQLiteCantOpenDatabaseException: Cannot open database '/storage/emulated/0/Submarine/Database/submarine-database.db' with flags 0x30000000: Directory /storage/emulated/0/Submarine/Database doesn't exist`
Even after creating the directories it couldn't create the db file.
After trying to request the `WRITE_EXTERNAL_STORAGE` permissions in the MainActivity, I stumbled upon [this](https://stackoverflow.com/questions/73620790/android-13-how-to-request-write-external-storage) thread which describes that with Android 13, you cannot access the external storage anymore and have to use the internal storage of the app.

So, in order to fix this, the `getDatabase` function in `AppDatabase.kt` would have to look like this:
```kotlin
fun getDatabase(context: Context): AppDatabase {
    if (INSTANCE == null) {
        synchronized(this) {
            INSTANCE = Room.databaseBuilder(
                context.applicationContext,
                AppDatabase::class.java,
                _dbFileName
            ).build()
        }
    }
    return INSTANCE!!
}
```

simondankelmann, 2023-05-11:
Thanks @d4ve10 for sharing your solution !

computerfreek274, 2023-05-12:
> I also had the same problem using Android 13 on the OnePlus 8 while testing the app. The error message from Android Studio is `android.database.sqlite.SQLiteCantOpenDatabaseException: Cannot open [...]

Thanks that still did not fix it. I also tried updating some of the Bluetooth permissions after trying your fix. Still just crashes I was able to get it to work on an older tablet but not very stable on that neither I believe that has Android 12 on it. I'll keep playing around me have to run the app through Android studio instead maybe I can update. I would like to get this to work I have a flipper zero however this would be a great extra tool to play with I would also like to get a touch screen working with it down the road that would be pretty cool. 

If we had a Wi-Fi interface to it probably would be a little more stable than the Bluetooth as well.

computerfreek274, 2023-05-12:
> I also had the same problem using Android 13 on the OnePlus 8 while testing the app. The error message from Android Studio is `android.database.sqlite.SQLiteCantOpenDatabaseException: Cannot open [...]

Fixed! Thank you right after writing that last reply I was able to re-compile it with your fix a few other Android 13 fixes I found online. The most important thing that needed to be done is allow all the permissions but then an Android 13 you need to search and settings and go find special access and Android 13 settings then click on off file access in there you need to turn on off all access for submarine.

closed 2023-05-12 by computerfreek274