simondankelmann/SubMarine #6: Compilation error - SPIFFS.h and BluetoothSerial.h

gorizilla, 2023-05-18:
The includes are not found in the .ino file.
Can you explain how to get it running?
---
SubMarine:2:10: fatal error: SPIFFS.h: No such file or directory
    2 | #include "SPIFFS.h"
---

simondankelmann, 2023-05-19:
Do you use an ESP32 or any other board ?

gorizilla, 2023-05-19:
Esp32 Wroom

simondankelmann, 2023-05-19:
so on my ubuntu machine the SPIFFS.h file is at:

~/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPIFFS/src/

and BluetoothSerial.h is at:

~/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/BluetoothSerial/src/

so i guess you need to install the ESP32 package by adding this line:

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

to Preferences->"Additional Boardmanager URLs" in your Arduino IDE

hope this helps :)

gorizilla, 2023-05-21:
So good so far. 
It was the wrong board.
But...
The app crashes often and the only thing that works is detect signal. 
Radar gets nothing.
Record gets nothing. 
Load .sub icon says import but nothing to see in the list and when I press send nothing happens. 
Do you have any clue? 
Maybe wiring is wrong or so on.
Please help. 

It would be really nice when we got an easy and cheap alternative to flipper zero subG.
My hope is an complete subg substitute for the flipper zero based on esp32 via web interface or Android app.

simondankelmann, 2023-06-02:
Sry for the late response.
Maybe it is due to the wiring, can you See any Output in the arduino serial Monitor regarding the cc1101 Connection?