simondankelmann/SubMarine #7: Sometimes hangs on transmitting anyway to add a reboot?

Nhscan, 2023-05-31:
Sometimes the unit hangs on transmitting you could tell that by the blue and red light stays at the same time. Anyway of adding a reboot option so you don't have to manually reboot the device?

simondankelmann, 2023-05-31:
i guess when it hangs on transmission mode, some error occured and most propably the device wouldnt listen to commands anyway. can you trigger this behaviour by maybe sending some special payload or does it just happen from time to time ?

Nhscan, 2023-06-02:
Seems random however I can recreate it by transmitting a few times to maybe 10 times drom the database nothing custom just stuff I had recorded. When the unit hangs it seems like it's still listening it doesn't show disconnected. However you can't really receive any transmissions and you can't transmit as well. Not a huge big deal I just hit the reset button and I was just wondering if there's a way to add a reset. Or maybe some kind of watchdog. 

Periscope also does that work very consistently for me sometimes it receives stuff and sometimes it doesn't doesn't matter what I'm transmitting in the area from remotes to a flipper zero it just doesn't pick up anything I can get it to work randomly but I'm not sure how I get it to work. 

Anyways excellent job on the app and fixing the Android 13 problem with permissions. Thanks. I really hope this project continues. Just wondering 
Why did you decide to go the Bluetooth route with an app and not with a web interface and connected to the unit with wifi? Just asking because my son waiting to use it but he has an iphone.

simondankelmann, 2023-06-02:
sry to hear that it doesnt work, i will try to reproduce the bug with my device. but as of now it seems to run prety smoothly. which boards exactly do you use ? 

i decided to go the bluetooth way because i didnt want those "This Wifi Connection has no Internet"-Messages and then many phones just switch to another wifi network. also MAYBE MAYBE if there are some users in future it could be nice to do some "signal" sharing etc over internet then

Nhscan, 2023-06-02:
I board I am using are these I never thought about it that it could be the boards Teyleten Robot D1 Mini ESP32 ESP-WROOM-32 WLAN WiFi+Bluetooth Internet of Things IoT Development Board for Arduino NodeMCU (3pcs) https://a.co/d/4vBIkLK 
What board gave you tested it with I actually have one unit built on a PCB board and one with just jumpers both act funny for me. It is possible it is my Samsung s22 ultra as well. It you have a board recommendation let me know I'll grab one and try it. 

I was just wondering why with the Bluetooth thing.  Maybe down the road someone will build an ios app. I can't stand ios so it doesn't matter to me. 😂 
That would be cool if down the road if you can utilize the wifi for like debug output so if it was connected with with you can monitor it via web interface. I don't know just thinking what the wifi chip can be used for. 

Thanks again!

simondankelmann, 2023-06-04:
The Board looks good. so it seems to be buggy somehow. have you checked if there is anything interesting in arduino serial monitor ?

Nhscan, 2023-06-04:
I have and not anything out of the usual nothing really looks out of place it does stop responding as you said viewing it on the serial side. So adding some kind of reset probably would not work like you said. It does seem to do it more if I retransmit a saved signal and then try to go to like Periscope or back to record signal when I go back to record signal it will not grab another signal it's just basically Frozen. I retested it with a new cc-1101 and that did not help I tested the cc-1101 on slipper zero and that's working fine with transmit and receive it did have a bad one not too long ago that would only receive and not transmit. 

Definitely something buggy on my end I don't know or including you see it I just can't get it to work right hopping around in the menu is definitely freezes it and makes it so I can't rerecord anything. I was trying to do a simple test tonight recording on 315 MHz and outlet I have and it still will not record the thing I've gotten it to record one other time and that's it really strange. Periscope in general just does not work for me I can get it to maybe work one time out of an hour of playing it with it. So that's why I'm wondering if it's my hardware. 

What are I just went back through all the pinouts just to verify that everything is connected and all good I built a second one using the same kind of board and  cc1101 wasn't just jumper wires and I get the same results. 

I really don't know if it's my phone or if it's my hardware at this point

Nhscan, 2023-06-07:
Okay tried two different boards and the first issue is that even without a cc1101 connected and if you go to record signal it's constantly recording a signal on whatever frequency you have set. With like a key or hash of like 3030 I'll include a picture so you can see what I'm talking about well that same issue is flooding my record signals and now I tried it on the board I showed you before and on a new wemos bored not the D1 the full size board and I have the same results. https://photos.app.goo.gl/vhHW1gAQGasuxwKo7

simondankelmann, 2023-06-08:
I See the values are all 32000 this seems to come from the Board then. 32000 is the configured timeout for a Signal in the arduino Code. Maybe its somehow relates to the Pin configuration. Gdo0 and gdo2 are for Receiving and sending Signals. If there constant voltage it will Always think it Receives Something

Nhscan, 2023-06-08:
I don't see this happening with testing radiolob or repeater using ELECHOUSE_CC1101. I do see and understand what you mean I am more of a hardware person not so much software. It is definitely strange that it open happens or that I only see it with this code with or without cc1101 connected. It is what i figured after it was happening without cc1101 wired up. So you are not getting this at all with any of your boards ? I tested on two totally different board types and brands still get it on both. I do not think a pull down resistor would help as it seems like it's more of an internal board problem. 
Is there a way we can filter it out via software? 

Also how could I lower the gain when on the record page? Can I just adjust it in the .into ? I was thinking maybe it's a gain think but what you are saying I guess that would not be the case. 

I did get it somewhat working better tonight however I do get bombarded with the false positive 32000 a lot more than I can record if I am lucky in between it but it is very tricky. I then thought well maybe it's a power thing so I went battery pack only still no charge so I could rule out a dirty power supply.

simondankelmann, 2023-06-08:
Did you Check If gdo0 and gdo02 are wired to the exact Pins defined in Code ? Inguess this cannot be filtered by Software. I dont geht this Errors on my Board. Which Frequency did you try ?

Nhscan, 2023-06-10:
Okay I could almost retreat the hanging up and it just going into a constant transmit mode where it's just almost like a jammer. If I record a signal and it doesn't detect the protocol and I hit retransmit unless every time maybe every other time it will go into a continuous transmit I could pick it up on the SDR and it actually will jam the signal trying to reach whatever I was originally recording.

rimoliga, 2023-07-13:
Hello, I'm also having ghost readings, no matter what pins I try for gdo0 and gdo2, even if they are disconected I get signals in 
the "detect signal" section. In periscope and record signal I get nothing.

Nhscan, 2023-07-14:
> Hello, I'm also having ghost readings, no matter what pins I try for gdo0 and gdo2, even if they are disconected I get signals in the "detect signal" section. In periscope and record signal I get nothing.

Building it on a PCB board and hot gluing over all the connectors seems to have insulated something because it seems to definitely be a lot better and my newer versions that I have built. I know the ones I built using jumper wires was a no-go keeping your wires short from point a to point b and instantly in it seems to definitely help with mine at least. I have two that work very well and a few others that don't another note. I also did not have any luck with the esp32 words I posted in a link I actually got the same boards that you would use on a flipper zero for Wi-Fi the esp32-s so much better super quiet haven't had much issues besides occasionally needing to reboot it for Bluetooth reasons. I meant to post an update on here and how it was going I would try different esp32 different brand different everything tried to get the 32s.