simondankelmann/SubMarine #8: Can't record signals, but detection works

elschopi, 2023-08-14:
My phone is a fully stock, up-to-date Samsung Galaxy S22 Ultra. The app works so far. I can connect to the ESP32 (AZ delivery NodeMCU board) and configure things.
My problems are as follows:
1) Sometimes when configuring the CC1101, it somehow reverts to another configuration. For example, I set it to 433.92 MHz FSK and after saving configuration and re-entering config screen it is configured to 900 MHz etc. 
2) I can't record signals. However, in the "Detection" screen, when I adjust the slider a bit, I can see signals being detected. Even for a remote control being near the device I need to adjust sensitivity, but that may be just my hardware. 

Adding a sensitivity slider (like the one in detection) in the record screen may solve the problem.

simondankelmann, 2023-08-15:
Hi, about Nr 1:

Is it possible that in between you replayed a signal or something? In that case the config would be set to the signals config.

Nr. 2
The slider is a good idea, I will keep that in mind for the future :) 

Thanks !

elschopi, 2023-08-15:
Hi, thanks for the fast feedback!
1) What I've done so far was switch between the screens, but didn't replay a signal (since I wasn't able to record a signal yet).

rimoliga, 2023-09-01:
Hi, I'm experiencing the same  behavior, I detect the signals from a key that I have when using detection mode, but nothing shows up in periscope or record signal. I tried this tool https://github.com/mcore1976/cc1101-tool and I was able to record raw packets with no problem using the same hardware. If there is any thing we can do to test this more efittiently I'm open to try anything you need.

Marcin-Kwiatkowski-NA, 2023-11-07:
@rimoliga  try this, it should help:
#define PIN_LED_ONBOARD 13
#define PIN_GDO0 2

o2809013, 2024-08-18:
#define PIN_GDO0 12
#define PIN_GDO2 27

works