simondankelmann/SubMarine #10: Error entering code esp32

Lamex9, 2023-11-03:
Hello. I'm actually doing it myself after reading it all and starting to load code for esp 32. Error "SubMarine.ino:74:38: fatal error:

ELECHOUSE_CC1101_SRC_DRV.h: No such file or directory"
I tried deleting it and got the error again 
"SubMarine.ino:9:17: error: variable or field 'btCallback' declared void"
I can't describe it exactly because I'm not very good at English. Do you have any better directions? Thank you

simondankelmann, 2023-11-03:
It seems you dont have the included libraries on your system.
please install or download this :

https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/tree/master

Lamex9, 2023-11-03:
Oh. I thought no one would respond. Thank you.
Currently I still have the error "btcallback"... Now how to solve?.
Currently I am using Esp 32 and NC 1101 8pin

Lamex9, 2023-11-19:
Everything is okay . Can run small bin files. Large files are not possible. Do you have any suggestions?

simondankelmann, 2023-11-19:
what do you mean with large ? possibly its just more than the esp can handle at once