simondankelmann/SubMarine #14: Bluetooth not found my esp32

wh1ter0z, 2023-12-06:
Hi 
I compiled and tested on my Samsung S10 (4g) (rooted) 

Apps runs well but my eso32 not found (all permission granted) 

In bluetooth i found it manualy but in android app never show ....

simondankelmann, 2023-12-08:
acutally settings button does not do anything at the moment :P
what android version is the phone that doesnt work please ?

wh1ter0z, 2023-12-08:
> acutally settings button does not do anything at the moment :P
> what android version is the phone that doesnt work please ?

ANDROID 11 (API 30) 
samsung s10 4g SM-G973F/DS

wh1ter0z, 2023-12-08:
> can you try this:
> 
> https://github.com/simondankelmann/SubMarine/blob/main/Android/SubMarine/app/debug/app-debug.apk
> 
> i changed something in the bleutooth permission request.

I will try now

wh1ter0z, 2023-12-08:
> can you try this:
> 
> https://github.com/simondankelmann/SubMarine/blob/main/Android/SubMarine/app/debug/app-debug.apk
> 
> i changed something in the bleutooth permission request.

Same , not show Submarine Bluetooth in app

simondankelmann, 2023-12-08:
hmm strange...

wh1ter0z, 2023-12-08:
> hmm strange...

Any idea to fix it?

simondankelmann, 2023-12-08:
can you maybe try to hook up your phone to the pc and run the app from android studio, maybe something is in the adb log

wh1ter0z, 2023-12-08:
> can you maybe try to hook up your phone to the pc and run the app from android studio, maybe something is in the adb log

Android studio is filtered but at night i can do it 

Now 15:05 (Iran)

wh1ter0z, 2023-12-08:
> let me know :)

I will 

If u have telegram please give me your id or u can text me @Thereisnomind

wh1ter0z, 2023-12-08:
[image]

Scan not granted

simondankelmann, 2023-12-08:
i changed it 20 minutes ago after your post :)

wh1ter0z, 2023-12-08:
> https://github.com/simondankelmann/SubMarine/commit/f749e52d9273574f6687d974fccdc55203581b01

Fixed sir ❤👍 

Thanks for help

simondankelmann, 2023-12-08:
nice :) thank you for the logs. maybe this helps some more people with sdk level 30 and below !

closed 2023-12-08 by simondankelmann