simondankelmann/SubMarine #18: Latest update and Samsung Galaxy s22 ultra android ver. 14

Nhscan, 2023-12-23:
Since the latest update this phone will not find the device via Bluetooth device is working fine able to connect from my wife's phone which is much older in my Samsung tablet which is much older running Android 13. I got all the permissions even special permissions still won't find anything any tips would be appreciated.

simondankelmann, 2024-01-19:
did you rename the bluetooth device in your arduino sketch by any chance ? the app got modified so it only shows devices named SubMarine becausse people had issues with many other bluetooth devices nearby

Nhscan, 2024-01-19:
Not at all but I will double check. I know I had a couple of updates pushed to my Samsung device like I said it still works on my tablet with Android 12 or 13 on it but on the Samsung Galaxy s22 ultra it's got the newest and greatest Android so I think that has something to do with it obviously. Plus I have set three separate boards made and each one was flashed at different times so I highly doubt I renamed any of it and probably re-downloaded the sketch a few times.

simondankelmann, 2024-01-20:
hmm strange thing but yes maybe something on the smartphone is the issue here

FLDSMDFR-lab, 2024-07-23:
Hello everyone,
I have almost the same problem. I have tested it with Android 11 (Samsung Galaxy A40). It recognizes “SubMarine” as a normal Bluetooth device, but not in the app. With another smartphone, which has Android 13 (Samsung Galaxy A22 5G), it works again in the app.
Is there by any chance the possibility to edit the .apk? or that you could make it available to us?

Greetings Jason