simondankelmann/SubMarine #19: Problem with Transmission

Paolo5466, 2024-01-19:
Hi.
First of all, thanks for this project, it is very interesting and fun to work with.

I'm experiencing a problem with transmission.
I'm using a NodeMCU ESP32-WROOM-32 (38 pin version, see image below) and an Ebyte ebyte E07 (M1101D) CC1101 8 pin antenna.

[image]

I'm using pin 2 as "PIN_LED_ONBOARD" and pin 12 as "PIN_GDO0"
I also tried to use pin 2 as "PIN_GDO0", but I have the same result.

From the serial monitor, everything seems to be working. I don't get any error and I see logs for both received and transmitted signals.

I can record new signals in the database, but when retransmitted they don't work.

Also I noticed that if I use the transmit function from the app, to be able to use "Periscope mode" or "Record signal" I have to reboot the ESP. Only "Detect Signal" is always working.
After a reboot Periscope and record work, till the next transmission. Then I have to reboot again.

I tested 2 antennas (same type) but nothing changed.

Could you kindly help?

If you need any log/additional info, I can provide them.

Thanks in advance.

PS: I couldn't find the TI-C1101 10 pin antenna that you used in the schematics.
Do you know any store that has that model available?

Bye

Paolo5466, 2024-01-19:
Hi
Thanks
I'm working with 433.92 MHz. at the moment.
I'll try changing pin for GDO0, as you suggested.

Regarding the receiving issue after using the transmission, do you have any reccomendation?
I suspect something in the code could be missing, switching from a function to another maybe the "CC1101_TX" variable doesn't get set back to false somewhere.

Thanks again

Bye

simondankelmann, 2024-01-19:
sadly i dont have any other recommendation then playing around with it and debug :D i think you might be right, it might be an issue with switching rx/tx. maybe its even a problem with the ebyte when switching the mode.