simondankelmann/SubMarine #23: Bluetooth not working

Thunderpebble, 2024-11-13:
I have made this project with the ESP32 D1 mini and the CC1101 (8 pin version), hence the wiring is somewhat different than the provided picture if I follow the provided pin to pin assignment.
[image]

The problem starts when I try to run the code from Arduino IDE; the program is compiled and loaded successfully, and keeps repeating the following code, but no Bluetooth device is visible.
[   482][I][BluetoothSerial.cpp:575] esp_bt_gap_cb(): ESP_BT_GAP_CONFIG_EIR_DATA_EVT: stat:0 num:4
[   483][I][BluetoothSerial.cpp:707] _init_bt(): device name set

I have tried the SerialToSerialBT example for the ESP32 D1 mini in the Arduino IDE and this works just fine (while seemingly using the same code)

Do you have any idea what could be going wrong? 
Thanks in advance for your help.

Thunderpebble, 2024-11-19:
EDIT: I figured out that the ESP32 is reset when the code gets to the ELECHOUSE_cc1101.Init(); command. This is likely due to an incorrect connection of the module.

simondankelmann, 2024-11-20:
Interesting, maybe you're right and there is something wrong in the schematic. If you figure out to get the board connected please let me know how so I can update the schematics or of course you could create a pull request if you like.

Thank you for your input!

Thunderpebble, 2024-11-28:
Hi Simon, I did get the board working using the schematic above. 
This means that the schematic you show is not correct for all ESP32-C1101 combinations.
Let me know if you want me to send you the Fritzing file.
One thing to keep in mind is that if you connect the blue line to IO2 on the ESP32, the code on line 76 should state #define PIN_GDO0 2.
In the end the main error that caused my device not to run was that the C1101 .h file was moved to a different folder, so the C1101 code could not be found and caused the board to keep resetting.

simondankelmann, 2024-11-28:
Thank you for your effort and sharing the solution ! would you like to create a pullrequest for that changes ?

simondankelmann, 2024-11-28:
Sure no pressure. In general the best way is you clone the current version into a new folder on your device, create a new branch, add you changes to it and then commit and push back to github. from there you can easily create a pullrequest :)

Thunderpebble, 2024-11-28:
Aah, I see. I hoped it would be as simple as comparing two .fzz files but it involves a bit more.