simondankelmann/SubMarine #24: Device doesn't boot at all

tegtheevi, 2025-04-26:
All wiring is good and all steps are done correctly but the device doesnt boot at all any help would be appreciated

simondankelmann, 2025-04-27:
Do you have some logs Out of the arduino Ide ?

tegtheevi, 2025-04-27:
> Do you have some logs Out of the arduino Ide ?
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
20:20:13.836 configsip: 0, SPIWP:0xee
20:20:13.838 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:20:13.844 mode:DIO, clock div:1
20:20:13.847 load:0x3fff0030,len:4660
20:20:13.849 load:0x40078000,len:15576
20:20:13.851 load:0x40080400,len:4
20:20:13.852 load:0x40080404,len:3152
20:20:13.856 entry 0x400805a0
20:20:15.322 ets Jul 29 2019 12:21:46
20:20:15.324 
20:20:15.324 rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
20:20:15.330 configsip: 0, SPIWP:0xee
20:20:15.332 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:20:15.338 mode:DIO, clock div:1
20:20:15.340 load:0x3fff0030,len:4660
20:20:15.343 load:0x40078000,len:15576
20:20:15.345 load:0x40080400,len:4
20:20:15.347 load:0x40080404,len:3152
20:20:15.349 entry 0x400805a0 
I tried flashing 2 esp32 wroom 32u but had the same result

tegtheevi, 2025-04-27:
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
20:20:13.836 configsip: 0, SPIWP:0xee
20:20:13.838 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:20:13.844 mode:DIO, clock div:1
20:20:13.847 load:0x3fff0030,len:4660
20:20:13.849 load:0x40078000,len:15576
20:20:13.851 load:0x40080400,len:4
20:20:13.852 load:0x40080404,len:3152
20:20:13.856 entry 0x400805a0
20:20:15.322 ets Jul 29 2019 12:21:46
20:20:15.324 
20:20:15.324 rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
20:20:15.330 configsip: 0, SPIWP:0xee
20:20:15.332 clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:20:15.338 mode:DIO, clock div:1
20:20:15.340 load:0x3fff0030,len:4660
20:20:15.343 load:0x40078000,len:15576
20:20:15.345 load:0x40080400,len:4
20:20:15.347 load:0x40080404,len:3152
20:20:15.349 entry 0x400805a0

simondankelmann, 2025-04-28:
Maybe there is a mistake in the Wiring, - please Look at issue #23 there was a similar issue. If this can be confirmed, maybe the Wiring Diagram in the repo has to be updated

tegtheevi, 2025-04-28:
Im using esp32 wroom 32u but when i flash the esp before connecting the wires i open the serial monitor and see the prints and i get the same error that i sent you and no other serial prints 
If i made a mistake please point it out